kevinlekiller/amdctl #3: Modify NB voltage AMD fam12h

vinibali, 2016-03-30:
Hello.

It's a great project. Are you planning to implement a function to modify the Llano APU's NB voltages? In FusionTweaker I can do this, but unfortunately in Linux there is no tool to do that, I'm also using TurionPowerControl at the same time, but the project seems like dead. If you have any idea or some code to test, than I have a desktop and mobile with the 1st gen AMD APU.

Thanks

kevinlekiller, 2016-04-05:
Hi @vinibali sorry for the late reply.

It is possible to do the NB voltages, the reason I left it out is because the implementation is quite different between the various families of AMD CPU's, although I can create a branch just for the 12h family for North Bridge voltages, I might do that in the next few weeks.

vinibali, 2016-04-05:
Hi @kevinlekiller.
Np, thanks for your time and effort!

kevinlekiller, 2016-04-05:
Hi @vinibali, I'm currently looking at the BIOS/Kernel dev guide for 12h.

If you look at the PDF: http://support.amd.com/TechDocs/41131.pdf on page 469, there is no mention of a writable field for NbVid, although if you look at the 10h manual for example http://support.amd.com/TechDocs/31116.pdf on page 429, there is a writable field for NbVid at bits 31:25.

I've just looked at most of the guides from 10h to 16h and only 10h mentions that writable NbVid field at 31:25.

It could be they forgot to include the information in those other CPU families.

Do you know if FusionTweaker has the source code published? We could check which bits it's working on and confirm if it's 31:25 like in 10h.

kevinlekiller, 2016-04-05:
I found the FusionTweaker source code, I looked over it quickly, it's a large project so it's hard to to follow, but I think it's falling back on 10h numbers if the CPU is not a 16h : https://github.com/Jeje2312/fusiontweaker/blob/master/FusionTweaker/PStateMsr.cs#L246

vinibali, 2016-04-06:
I've fount the code.google variant too.
[https://fusiontweaker.googlecode.com/svn/trunk/](url)
This program was definately able to set the nb voltages!

kevinlekiller, 2016-04-06:
I'm looking at the amdctl code, it seems I've already implemented changing the voltage for the north bridge (-n    Set north bridge voltage id (vid).), I can't see any code blocking 12h CPU's from doing it also.

Did you mean the north bridge fid / did maybe? I think the issue with those is how much it varies between the CPU families.

vinibali, 2016-04-06:
Yes, the amdctl was able to show some value from the NBvid. So the program was able to print some data in the past too, but seems like the value is false. I was not enought brave to execute any command to change the NBvid.

sudo amdctl -g
Voltage ID encodings: SVI (serial)
Detected CPU model 1h, from family 12h with 4 CPU cores.

Core 0 | P-State Limits (non-turbo): Highest: 1 ; Lowest 7 | Current P-State: 2
 Pstate Status CpuFid CpuDid CpuVid CpuMult CpuFreq CpuVolt NbVid NbVolt IddVal IddDiv CpuCurr CpuPower
      0      1     12      0     13  28.00x 2800MHz  1388uV     0 1550uV    168     10  16.80A   23.31W
      1      1      9      0     20  25.00x 2500MHz  1300uV     0 1550uV    125     10  12.50A   16.25W
      2      1      7      0     23  23.00x 2300MHz  1262uV     0 1550uV    111     10  11.10A   14.01W
      3      1      5      0     25  21.00x 2100MHz  1238uV     0 1550uV    101     10  10.10A   12.50W
      4      1      3      0     28  19.00x 1900MHz  1200uV     0 1550uV     92     10   9.20A   11.04W
      5      1      8      1     32  16.00x 1600MHz  1150uV     0 1550uV     79     10   7.90A    9.09W
      6      1      8      2     37  12.00x 1200MHz  1088uV     0 1550uV     63     10   6.30A    6.85W
      7      1      8      3     43   8.00x  800MHz  1012uV     0 1550uV     48     10   4.80A    4.86W
current      1      7      0     23  23.00x 2300MHz  1262uV    44 1000uV

kevinlekiller, 2016-04-06:
Yeah that seems off, 12h CPU's might have a different way to calculate the voltage, I'll investigate, thanks!

kevinlekiller, 2016-04-06:
I've done some tests on a 15h someone gave me access to which is similar to 12h, in that the registers for the individual Pstates (0xc0010064 and up) have no NB vid field and trying to use the same field bits from the "current" Pstate results in getting no data (hence the 0 and 1550uV).

I figured, what if I try changing the current Pstate and getting the NB vid from the current Pstate. That did not work because when I change the current Pstate, it changes back instantly (because of AMD Cool n Quiet) and I have no time to get or set the data for the intended Pstate, and trying to force that Pstate by setting the min/max Pstate is not possible since those fields are read only (page 468 in 12h guide).

I'm not sure what the other tools are doing to set the voltage for individual Pstates, I went through fusiontweaker's source code a few times but it's very hard to follow. My guess is they found the right bits to set, which are undocumented in the AMD kernel and bios dev guides.

I found another tool called k15ctl https://github.com/tud-zih-energy/k15ctl which I tried on the 15h CPU I have access to, but it also shows 0 and 1.55v for the NB, so I assume he made the same mistake I did.

The mistake was, when I was going through the guides months ago, I assumed the NbVid field was missing from the individual Pstates (MSRC001_00[6B:64]) because I thought the person who wrote it forgot to put it in the guide, since the field is in the COFVID Status Register.

For now I will disable the -n switch for non 10h CPU's, as I don't have a Windows AMD computer I can't compile / debug fusiontweaker to see what exactly it's doing (which bits it's trying to set, maybe someone else can figure that out for us and submit a PR in the future?).

vinibali, 2016-04-07:
I've done a few test.
Normally my A8-3820 is running with 900Mhz NB. But the time MSR from Aida64 and the log from Fusiontweaker was saved, while the bios used its default settings.
Default:
Pstates Monitor screenshot
[https://www.dropbox.com/s/vfsn981q7mp695f/bios_def.jpg?dl=0](url)
Fusiontweaker dump side, not textbox:
[https://www.dropbox.com/s/wyf8lh3luvt5nqj/fusion_def.png?dl=0](url)
Aida64 MSR read:
------[ Versions ]------

Program Version : AIDA64 v5.70.3800
BenchDLL Version: 4.2.671-x64

------[ CPU Info ]------

CPU Type          : QuadCore AMD A8-3820, 2500 MHz (25 x 100)
CPU Alias         : Llano
CPU Platform      : Socket FM1
CPU Stepping      : LN-B0
Instruction Set   : x86, x86-64, MMX, 3DNow!, SSE, SSE2, SSE3, SSE4A
CPUID Manufacturer: AuthenticAMD
CPUID CPU Name    : AMD A8-3820 APU with Radeon(tm) HD Graphics
CPUID Revision    : 00300F10h
AMD K1x Brand ID  : 2951h  (A8-3820)
Platform ID       : D9h  (Socket FM1)
HTT / CMP Units   : 0 / 4
Max. NUMA Node    : 0

Tjmax Temperature    : 0 Celsius
HTC Temperature Limit: 80 Celsius

------[ Logical CPU #0 ]------

allcpu: Package 0 / Core 0 / Thread 0: Valid

CPUID 00000000: 00000006-68747541-444D4163-69746E65 [AuthenticAMD]
CPUID 00000001: 00300F10-00040800-00802009-178BFBFF
CPUID 00000002: 00000000-00000000-00000000-00000000
CPUID 00000003: 00000000-00000000-00000000-00000000
CPUID 00000005: 00000040-00000040-00000003-00000000
CPUID 00000006: 00000000-00000000-00000001-00000000
CPUID 80000000: 8000001B-68747541-444D4163-69746E65 [AuthenticAMD]
CPUID 80000001: 00300F10-20002951-000037FF-EFD3FBFF
CPUID 80000002: 20444D41-332D3841-20303238-20555041 [AMD A8-3820 APU ]
CPUID 80000003: 68746977-64615220-286E6F65-20296D74 [with Radeon(tm) ]
CPUID 80000004: 47204448-68706172-00736369-00000000 [HD Graphics]
CPUID 80000005: FF30FF10-FF30FF20-40020140-40020140
CPUID 80000006: 20800000-44004200-04008140-00000000
CPUID 80000007: 00000000-00000000-00000000-000003F9
CPUID 80000008: 00003028-00000000-00002003-00000000
CPUID 80000009: 00000000-00000000-00000000-00000000
CPUID 8000000A: 00000001-00000040-00000000-0000040F
CPUID 8000000B: 00000000-00000000-00000000-00000000
CPUID 8000000C: 00000000-00000000-00000000-00000000
CPUID 8000000D: 00000000-00000000-00000000-00000000
CPUID 8000000E: 00000000-00000000-00000000-00000000
CPUID 8000000F: 00000000-00000000-00000000-00000000
CPUID 80000010: 00000000-00000000-00000000-00000000
CPUID 80000011: 00000000-00000000-00000000-00000000
CPUID 80000012: 00000000-00000000-00000000-00000000
CPUID 80000013: 00000000-00000000-00000000-00000000
CPUID 80000014: 00000000-00000000-00000000-00000000
CPUID 80000015: 00000000-00000000-00000000-00000000
CPUID 80000016: 00000000-00000000-00000000-00000000
CPUID 80000017: 00000000-00000000-00000000-00000000
CPUID 80000018: 00000000-00000000-00000000-00000000
CPUID 80000019: F0300000-60100000-00000000-00000000
CPUID 8000001A: 00000003-00000000-00000000-00000000
CPUID 8000001B: 000000FF-00000000-00000000-00000000
CPUID 8FFFFFFF: 00000000-00000000-00000000-00000000
[... 411 lines cut ...]
MSR C0010069: 8000-014F-0000-4081 [16.00x] [1.1500 V] [ 7.90 A] [PState P4]
MSR C001006A: 8000-013F-0000-4A82 [12.00x] [1.0875 V] [ 6.30 A] [PState P5]
MSR C001006B: 8000-0130-0000-5683 [ 8.00x] [1.0125 V] [ 4.80 A] [PState P6]
MSR C0010070: 0000-0000-0007-0000
MSR C0010071: 0029-606F-6807-5683 [ 8.00x] [1.0125V]
MSR C0010071: 0029-606F-6807-5683 [ 8.00x] [1.0125V]
MSR C0010071: 0029-606F-6807-5683 [ 8.00x] [1.0125V]
MSR C0010071: 0029-606F-6807-5683 [ 8.00x] [1.0125V]
MSR C0010071: 0029-606F-6807-5683 [ 8.00x] [1.0125V]
MSR C0010140: 0000-0000-0000-0004
MSR C0010141: 0000-0000-0000-0000
MSR C0011023: 0000-0000-1020-0020

PerformanceFrequency = 2441416

CPU Clock M.Method = 6  (APIC K10)
PM Timer Address   = 2056  (0808h)

APIC Clock = 200.00 MHz  (AT = 104830 / TT = 15010 / try = 1 / SKL = 0.00 MHz)
APIC Clock = 200.01 MHz  (AT = 104838 / TT = 15010 / try = 1 / SKL = 0.00 MHz)
APIC Clock = 200.01 MHz  (AT = 104836 / TT = 15010 / try = 1 / SKL = 0.00 MHz)
APIC Clock = 200.00 MHz  (AT = 104848 / TT = 15012 / try = 1 / SKL = 0.00 MHz)
APIC Clock = 200.00 MHz  (AT = 104846 / TT = 15012 / try = 1 / SKL = 0.00 MHz)

Meanwhile I asked the developer Sven Wittek to join, will see :)

kevinlekiller, 2016-04-07:
It looks from those screenshots like the northbridge have their own P-states.

These are described on page 318: https://support.amd.com/TechDocs/41131.pdf

But they are not model specific registers, according to page 148 these are "PCI-defined configuration space" registers, which I have no experience with.

kevinlekiller, 2016-04-07:
Hi @vinibali, I've been reading more about the PCI space registers, I've been testing on a AMD A10-7870k, I got this so far:

Detected CPU model 38h, from family 15h with 4 CPU cores.
NB Pstate 0:  900uV 1200MHz
NB Pstate 1:  800uV 1066MHz
NB Pstate 2:  700uV 933MHz
NB Pstate 3:  575uV 733MHz

I'm not sure if that is correct, because I can't find any information on the a10-7870k for northbrige voltage / clock speed, although those numbers seem low to me.

I got this information by accessing the data on the filesystem itself, "/sys/devices/pci0000:00/0000:00:18.5/config"

D18F5x160 is what the AMD guide says P-state 0 is, D18 is the the PCI device, F5 is the function and x160 is hexadecimal for the address where the data is.

I could make a branch to read that data for 12h so we can compare to your logs above, I'll report back when I do that.

kevinlekiller, 2016-04-07:
I've adjusted my math for the clock speed, it seems better, I did find a screenshot of a 7850k which matches at least 1 of the clock speeds:

http://cdn.overclock.net/7/71/900x900px-LL-71c07b3e_Beast.jpeg

NB Pstate 0:  900uV 1800MHz
NB Pstate 1:  800uV 1600MHz
NB Pstate 2:  700uV 1400MHz
NB Pstate 3:  575uV 1100MHz

I think the math for the voltage is different than CPU's, I'll have to dig more.

Hopefully I can find more info for the 7850k/7870k/7890k on NB p-states to confirm numbers.

kevinlekiller, 2016-04-08:
So it was indeed a difference in math for the NB voltage on 15h (which is poorly documented).

I was able to confirm my numbers using this tool: https://github.com/johkra/amdmsrtweaker-lnx

Here's with the updated math:
NB Pstate 0: 1.225uV 1800MHz
NB Pstate 1: 1.175uV 1600MHz
NB Pstate 2: 1.125uV 1400MHz
NB Pstate 3: 1.062uV 1100MHz

Currently the NB code is just for the 7870k processor I have access to, so I will need to rewrite it to be more modular and support 12h and the other families.

vinibali, 2016-04-08:
Hi @kevinlekiller.
You've done such huge research :)
When I was reading the message what git forwards to my email account, than I've started to search about the 7870K NBstates, so the last math seems like correct:
Aida64 PState monitor:
[http://prohardver.hu/dl/cnt/2015-07/120369/picz/7870K_pstates.png](url)

kevinlekiller, 2016-04-08:
I'm now reading the 12h manual for NB vid's.

https://support.amd.com/TechDocs/41131.pdf

Reading seems simple, there's 2 addresses for 2 P-states

Pstate 0 is at page 318 (D18F3xDC Clock Power/Timing Control 2 Register)
NbPs0Vid is the field name for the Vid.

Pstate 1 is at page 350 (D18F6x90 NB P-state Config Low)
NbPs1Vid for the field name.

The complex part seems to be writing.

It says we need to set D18F6x90[NbPsCtrlDis] to 1, and that says D18F6x90[NbPsLock] needs to be 1 also.

Once both those are 1, then the Vid can be changed, then it says we need to wait D18F3xD8[VSRampSlamTime] before resetting D18F6x90[NbPsCtrlDis], which I guess means D18F6x90[NbPsLock] needs to be reset also after.

kevinlekiller, 2016-04-08:
I will work on adding code to read the 12h NB vid's first so we can confirm if they match what you posted in Aida64, then I'll work on the code to set it.

kevinlekiller, 2016-04-08:
I pushed the changes to read the 12h NB p-state vids to a branch, you can download here: https://github.com/kevinlekiller/amdctl/archive/12h_NbVid.zip

vinibali, 2016-04-10:
Sorry for the late reply.
The NB vid read seems like correct now.
P-State 0: 36 (vid), 1100mV
P-State 0: 44 (vid), 1000mV.

kevinlekiller, 2016-04-11:
Thanks, I will merge that branch and start working on being able set the vid's.