kevinlekiller/amdctl #10: How to use it?

alsoijw, 2019-03-30:
There is some manual? I want reduce voltage on cpu. As I understand I have to do stress test. How can I check some pstate? When I run `stress --cpu 8` I can see like that
```
Voltage ID encodings: SVI (serial)
Detected CPU model 65h, from family 15h with 4 CPU cores.

Core 0 | P-State Limits (non-turbo): Highest: 1 ; Lowest 5 | Current P-State: 1
 Pstate Status CpuFid CpuDid CpuVid CpuMult CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1     20      0     28  18.00x 3600MHz  1200mV    145     10  14.50A   17.40W
      1      1     16      0     54  16.00x 3200MHz   875mV    107     10  10.70A    9.36W
      2      1     13      0     70  14.50x 2900MHz   675mV     87     10   8.70A    5.87W
      3      1     11      0     78  13.50x 2700MHz   575mV     77     10   7.70A    4.43W
      4      1      7      0     96  11.50x 2300MHz   350mV     58     10   5.80A    2.03W
      5      1     20      1    110   9.00x 1800MHz   175mV     42     10   4.20A    0.73W
current      1     11      0     85  13.50x 2700MHz   488mV
```
What state I have to edit? How to switch to another state?

echaskaris, 2019-06-14:
You modify the CpuVid field. Higher is lower voltage.

wmarkow, 2019-07-01:
I have a few questions related to the output from @alsoijw:
 * ```P-State Limits (non-turbo): Highest: 1 ; Lowest 5```
the non turbo P-states are from highest **1** to lowest **5**. However below there are six states. The first one with **0** id - I assume that this is the **turbo** one, right?
 * the first line ```Current P-State: 1``` says that current P-State is with number **1** (or this 1 means something else?), however at the bottom it says 
```current      1     11      0     85  13.50x 2700MHz   488mV```
which can give a clue - by looking at the 2700MHz frequency - that the current state is **3**. In which P-State is current?

wmarkow, 2019-07-02:
I think I know what is going on, related to my question. It looks like **amdctl** doesn't correctly distinguish between **turbo** and **non-turbo** P-States. In the example from @alsoijw, his processor has probably three **turbo** states and four  **non-turbo** and it should display like:
```
Core 0 | P-State Limits (non-turbo): Highest: 2 ; Lowest 6 | Current P-State: 1
```
More info in #9

alsoijw, 2019-07-06:
I tried to reduce voltage, so I got some voltage value. I could open gedit and typed some text. But when I open video in firefox without hardware acceleration system crash, so I got black screen. As I remember stress utility continued work, non firefox.

closed 2022-03-13 by kevinlekiller