kevinlekiller/amdctl #9: Is pstates list correct?

alsoijw, 2019-03-30:
I have AMD Quad Core A12-9720P
```
$ cat /proc/cpuinfo 
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 21
model		: 101
model name	: AMD A12-9720P RADEON R7, 12 COMPUTE CORES 4C+8G
stepping	: 1
microcode	: 0x6006118
cpu MHz		: 1635.759
cache size	: 1024 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 2
apicid		: 16
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good acc_power nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs xop skinit wdt lwp fma4 tce nodeid_msr tbm topoext perfctr_core perfctr_nb bpext ptsc mwaitx cpb hw_pstate ssbd vmmcall fsgsbase bmi1 avx2 smep bmi2 xsaveopt arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif overflow_recov
bugs		: fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 5391.99
TLB size	: 1536 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 48 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate cpb eff_freq_ro acc_power [13]

processor	: 1
vendor_id	: AuthenticAMD
cpu family	: 21
model		: 101
model name	: AMD A12-9720P RADEON R7, 12 COMPUTE CORES 4C+8G
stepping	: 1
microcode	: 0x6006118
cpu MHz		: 1588.258
cache size	: 1024 KB
physical id	: 0
siblings	: 4
core id		: 1
cpu cores	: 2
apicid		: 17
initial apicid	: 1
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good acc_power nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs xop skinit wdt lwp fma4 tce nodeid_msr tbm topoext perfctr_core perfctr_nb bpext ptsc mwaitx cpb hw_pstate ssbd vmmcall fsgsbase bmi1 avx2 smep bmi2 xsaveopt arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif overflow_recov
bugs		: fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 5391.99
TLB size	: 1536 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 48 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate cpb eff_freq_ro acc_power [13]

processor	: 2
vendor_id	: AuthenticAMD
cpu family	: 21
model		: 101
model name	: AMD A12-9720P RADEON R7, 12 COMPUTE CORES 4C+8G
stepping	: 1
microcode	: 0x6006118
cpu MHz		: 1672.267
cache size	: 1024 KB
physical id	: 0
siblings	: 4
core id		: 2
cpu cores	: 2
apicid		: 18
initial apicid	: 2
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good acc_power nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs xop skinit wdt lwp fma4 tce nodeid_msr tbm topoext perfctr_core perfctr_nb bpext ptsc mwaitx cpb hw_pstate ssbd vmmcall fsgsbase bmi1 avx2 smep bmi2 xsaveopt arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif overflow_recov
bugs		: fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 5391.99
TLB size	: 1536 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 48 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate cpb eff_freq_ro acc_power [13]

processor	: 3
vendor_id	: AuthenticAMD
cpu family	: 21
model		: 101
model name	: AMD A12-9720P RADEON R7, 12 COMPUTE CORES 4C+8G
stepping	: 1
microcode	: 0x6006118
cpu MHz		: 1659.357
cache size	: 1024 KB
physical id	: 0
siblings	: 4
core id		: 3
cpu cores	: 2
apicid		: 19
initial apicid	: 3
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good acc_power nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs xop skinit wdt lwp fma4 tce nodeid_msr tbm topoext perfctr_core perfctr_nb bpext ptsc mwaitx cpb hw_pstate ssbd vmmcall fsgsbase bmi1 avx2 smep bmi2 xsaveopt arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif overflow_recov
bugs		: fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 5391.99
TLB size	: 1536 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 48 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate cpb eff_freq_ro acc_power [13]
```
Minimal supported frequence is 1400000. Max turbo core is 3600000.
```
$ cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies 
2700000 2300000 1800000 1400000
```
But amdctl says that lowest state is 5. But 5 is 
```
 sudo amdctl -g -c0
[sudo] пароль для alsoijw: 
Voltage ID encodings: SVI (serial)
Detected CPU model 65h, from family 15h with 4 CPU cores.

Core 0 | P-State Limits (non-turbo): Highest: 1 ; Lowest 5 | Current P-State: 4
 Pstate Status CpuFid CpuDid CpuVid CpuMult CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1     20      0     28  18.00x 3600MHz  1200mV    145     10  14.50A   17.40W
      1      1     16      0     54  16.00x 3200MHz   875mV    107     10  10.70A    9.36W
      2      1     13      0     70  14.50x 2900MHz   675mV     87     10   8.70A    5.87W
      3      1     11      0     78  13.50x 2700MHz   575mV     77     10   7.70A    4.43W
      4      1      7      0     96  11.50x 2300MHz   350mV     58     10   5.80A    2.03W
      5      1     20      1    110   9.00x 1800MHz   175mV     42     10   4.20A    0.73W
current      1     12      1    118   7.00x 1400MHz    75mV
Northbridge:
P-State 0: 104 (vid),   250mV, 1100MHz
P-State 1: 108 (vid),   200mV, 1000MHz
P-State 2: 108 (vid),   200mV, 1000MHz
P-State 3: 124 (vid),     0mV, 700MHz
```
I can manually get information about 6 state.
```
 sudo amdctl -g -c0 -p6
Voltage ID encodings: SVI (serial)
Detected CPU model 65h, from family 15h with 4 CPU cores.

Core 0 | P-State Limits (non-turbo): Highest: 1 ; Lowest 5 | Current P-State: 3
 Pstate Status CpuFid CpuDid CpuVid CpuMult CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      6      1     12      1    118   7.00x 1400MHz    75mV     32     10   3.20A    0.24W
current      1     20      1     70   9.00x 1800MHz   675mV
Northbridge:
P-State 0: 104 (vid),   250mV, 1100MHz
P-State 1: 108 (vid),   200mV, 1000MHz
P-State 2: 108 (vid),   200mV, 1000MHz
P-State 3: 124 (vid),     0mV, 700MHz
```
If I continue iterate states I will get
```
$  sudo amdctl -g -c0 -p7
Voltage ID encodings: SVI (serial)
Detected CPU model 65h, from family 15h with 4 CPU cores.

Core 0 | P-State Limits (non-turbo): Highest: 1 ; Lowest 5 | Current P-State: 4
 Pstate Status CpuFid CpuDid CpuVid CpuMult CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      7      0      0      1      0   4.00x  800MHz  1550mV    202    100   2.02A    3.13W
current      1     12      1    110   7.00x 1400MHz   175mV
Northbridge:
P-State 0: 104 (vid),   250mV, 1100MHz
P-State 1: 108 (vid),   200mV, 1000MHz
P-State 2: 108 (vid),   200mV, 1000MHz
P-State 3: 124 (vid),     0mV, 700MHz
```
I don't know is it 7 state correct or not.

wmarkow, 2019-07-01:
@alsoijw, those AMD processors have a possibility to have at most 8 P-States, however not all of them may be available, it depends on the specific processor itself. It looks like **amdctl** tool is able to read all of them, no matter if they are valid.

In your example the state 7 is not valid because its **Status** is **0**. I also think that there is a small bug in **amdctl**. If you execute **amdctl -x** then it shows the explanation of the fields and for status we have:
```
Status:      If the P-State is enabled (1) or disabled (0).
```
however this message is wrong. According to the specs for [14H family](https://www.amd.com/system/files/TechDocs/43170_14h_Mod_00h-0Fh_BKDG.pdf) and [15H family models 60h-6Fh](https://www.amd.com/system/files/TechDocs/50742_15h_Models_60h-6Fh_BKDG.pdf) the meaning of this field is:
```
PstateEn. Read-write. 1=The P-state specified by this MSR is valid. 0=The P-state specified by this MSR is not valid. The purpose of this register is to indicate if the rest of the P-state information in the register is valid after a reset; it controls no hardware.
```

wmarkow, 2019-07-01:
I'm not quite sure if **amdctl** correctly distinguish between **non-turbo** and **turbo** P-States. Probably it silently assumes that there is olways one turbo state, which may be wrong. Some processors may have 0, 1 or more turbo states.
According to specs, the **D18F4x15C** register contains the number of turbo (or boosted) states but this registers is never read by **amdctl**.

wmarkow, 2019-07-02:
How to read prom PCI devices, for example **D18F4x15C** I have described in #10.

closed 2022-03-25 by kevinlekiller