kevinlekiller/amdctl #11: Disabling pstates doesn't work on A8-4500M

xCuri0, 2019-05-31:
On Windows I use PSCheck to disable all pstates except 2.3ghz to lock it to boost. I tried using this to do it on Linux but disabling does nothing

kevinlekiller, 2019-06-02:
I'm not sure it's been a while since I worked on this, have you tried this ? https://github.com/turionpowercontrol/tpc I think it supports bulldozer.

xCuri0, 2019-06-03:
@kevinlekiller that also doesn't work

xCuri0, 2019-06-03:
do you know how i can dump the MSRs on Windows ? I could then check the diff and see what register it edits

kevinlekiller, 2019-06-03:
Possibly with this: https://github.com/cocafe/msr-utility

Page 568 has info on bulldozer p-states: https://www.amd.com/system/files/TechDocs/42301_15h_Mod_00h-0Fh_BKDG.pdf#G5.3280729

xCuri0, 2019-06-03:
what wrmsr command would have i to use to set the minimum pstate to 1 (2.3ghz)

kevinlekiller, 2019-06-03:
I'm not sure, it's bit 63 of MSRC001_0064 according to the manual, so you'd need to read that msr, store it in a variable, then modify bit 63 to 0 in that variable and write the data in the variable to that msr.

xCuri0, 2019-06-06:
dunno what command i would have to use to on ```rdmsr``` to read that

xCuri0, 2019-06-06:
@kevinlekiller ```sudo rdmsr 0xc0010064 -f 63:63``` returns ```1```