kevinlekiller/amdctl #12: Can't set fid higher than 47?

Laaas, 2019-06-02:
Currently for pstate 0 my fid is 136, but when I manually set it I can't even set it above 47!
I am using a Ryzen 5 1600.

kevinlekiller, 2019-06-02:
It's possible they changed the max to be over 47, I haven't seen anything in the manual https://developer.amd.com/wp-content/resources/56255_3_03.PDF page 130ish

If you know the limit you can change this line https://github.com/kevinlekiller/amdctl/blob/master/amdctl.c#L117

Laaas, 2019-06-06:
I am not sure if this is relevant to this particular issue, but I patched the limits away with this patch:
```patch
diff -ruN a/amdctl.c b/amdctl.c
--- a/amdctl.c	2019-06-02 14:27:38.934074466 +0200
+++ b/amdctl.c	2019-06-03 13:52:39.681575900 +0200
@@ -114,9 +114,6 @@
 				break;
 			case 'f':
 				fid = atoi(optarg);
-				if (fid > 0x2f || fid < 0) {
-					error("Option -f must be a number 0 to 47");
-				}
 				break;
 			case 'n':
 				if (cpuFamily > AMD11H) {
@@ -395,6 +392,7 @@
 			NB_VID_BITS = "31:24";
 			break;
 		case AMD17H:
+			DIDS = 16;
 			CPU_VID_BITS = "21:14";
 			CPU_DID_BITS = "13:8";
 			CPU_FID_BITS = "7:0";
```

It's probably not entirely correct, since I couldn't find anything about the limits either, but this made it possible to set the pstate values to the ones reported by amdctl. Now, however, when I set the values, amdctl *will* report the values I set it to when I check them, but they do not seem to have any effect. I check my frequency with cpupower, but it seems to be the same, and I can't increase my maximum frequency above the old clock speed. Am I doing something wrong or is this a bug in amdctl?

kevinlekiller, 2019-06-06:
It might be a bug, from what I can see, the data is read from the MSR here: https://github.com/kevinlekiller/amdctl/blob/master/amdctl.c#L243

Then the data is modified in the next lines, and it is written back to the MSR here: https://github.com/kevinlekiller/amdctl/blob/master/amdctl.c#L260

And what is printed onto the terminal is from the modified buffer from line 243 : https://github.com/kevinlekiller/amdctl/blob/master/amdctl.c#L262

So maybe adding a read of the MSR again before line 262, then you can see if the print shows changed information.

Laaas, 2019-06-09:
So I made sure to check that the MSR contents are correct, and the P-State is correctly set, so amdctl is not buggy. However, it still does not seem to have any effect. I tried downclocking my CPU, but lscpu would report frequencies above this! Do you have any idea why this might be?

kevinlekiller, 2019-06-09:
I'm not sure, with Zen they changed/removed a lot of stuff and made the developer documentation not as good, possibly there's something else that needs to be done to make it work? 

Have you tried the AMD CBS / Zen Common options in your UEFI if you have those options? I believe that section allows changing P-States.

kevinlekiller, 2021-08-14:
Fixed with https://github.com/kevinlekiller/amdctl/commit/02eb745c115a702b634b636688c6b3be514e1754

closed 2021-08-14 by kevinlekiller