kevinlekiller/amdctl #13: Make fails - undefined reference to 'pow'

echaskaris, 2019-06-12:
$ make
gcc -Wall -pedantic -Wextra -std=c99 -O2 -lm    amdctl.c   -o amdctl
/usr/bin/ld: /tmp/ccAmmJxI.o: in function `northBridge':
amdctl.c:(.text+0x10ce): undefined reference to `pow'
collect2: error: ld returned 1 exit status
make: *** [<builtin>: amdctl] Error 1

echaskaris, 2019-06-13:
Putting -lm at the end of the gcc line fixes it.

kevinlekiller, 2019-06-14:
Not seeing the issue on my end, maybe we have different linker versions and it's parsing the options differently?

hyperfraise, 2019-12-14:
Same for me 
CPU 1900X
Ubuntu 18.04 LTS
gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04.1)

hyperfraise, 2019-12-14:
This command works though 

```
gcc -Wall -pedantic -Wextra -std=c99 -O2 amdctl.c   -o amdctl -lm
```

cmpute, 2020-01-06:
Confirm the problem, I also need to put -lm at the end

ulfw2020, 2020-11-08:
I'm also having this issue now, I've never had it before. I'm on Pop_OS!, with the latest versions of make, gcc etc. and linux kernel 5.8 if that's relevant. I previously used it without issue on Solus and Manjaro. I've checked the file CMakeLists.txt and the line that @hyperfraise put is there, without the word gcc. What's wrong? Any help is much appreciated.

echaskaris, 2020-11-08:
gcc -Wall -pedantic -Wextra -std=c99 -O2 amdctl.c   -o amdctl -lm
Try running this in terminal @ulfw2020

ulfw2020, 2020-11-08:
> gcc -Wall -pedantic -Wextra -std=c99 -O2 amdctl.c -o amdctl -lm
> Try running this in terminal @ulfw2020

Thanks for your response, I tried it, here's what it returned:
gcc -Wall -pedantic -Wextra -std=c99 -O2 amdctl.c -o amdctl -Im

/usr/bin/ld: /tmp/ccNRcSw4.o: in function `northBridge':
amdctl.c:(.text+0x1196): undefined reference to `pow'
collect2: error: ld returned 1 exit status

echaskaris, 2020-11-08:
[amdctl.zip](https://github.com/kevinlekiller/amdctl/files/5506848/amdctl.zip)
Here is my binary

ulfw2020, 2020-11-08:
Thanks, how do I use this? Forgive me for my incompetence haha.

echaskaris, 2020-11-08:
Np, run it as you would run an executable: ./amdctl This shows a help page
You will want to run it as this to get information about clock speeds and voltages: sudo ./amdctl -g 
 (probably you will need to make the file executable, I think chmod +x amdctl will work or Right click on your script and choose Properties -> Permissions -> Allow executing file as program)

ulfw2020, 2020-11-08:
Oh yeah, I forgot about that. Working now. Thanks for your help.

ulfw2020, 2020-11-08:
Shoot, while you're here. How do I make the changes permanent? I can set them using the commands, but when I restart they reset.

echaskaris, 2020-11-08:
:) In Ubuntu I had to play with services I believe, In other distros it is something else, google how to execute a script at startup

ulfw2020, 2020-11-08:
Mint, thanks.

echaskaris, 2020-11-08:
[undervoltScript.zip](https://github.com/kevinlekiller/amdctl/files/5506923/undervoltScript.zip)
Here is my script. Good luck

ulfw2020, 2020-11-08:
Thanks, gave yours a try, but it keeps saying the msr module isn't loaded. I've tried adding sudo modprobe msr to the script, but it gives me the same error. Any ideas? Also, is the best method of running the script on boot just moving it to /etc/init.d? Thanks.

echaskaris, 2020-11-08:
Google how to load module on startup i guess, I think for me it is loaded automatically

ulfw2020, 2020-11-08:
Managed to sort that issue by adding msr to /etc/modules, but now the script doesn't want to load on boot. It works fine when I run it manually using ./undervoltScript. I moved it to /etc/init.d. How do you get yours to load on boot? Thanks for the help. @echaskaris

echaskaris, 2020-11-08:
I didn't use init.d, can't help :P

kevinlekiller, 2020-11-08:
Most distros use systemd these days, if yours does, you can create a systemd unit file to run the script.

echaskaris, 2020-11-08:
@ulfw2020 First you want to find the values for your cpu, to know what values are stable

ulfw2020, 2020-11-08:
Thanks, that part's all sorted. Managed to undervolt from a max of 12.81W to 6.25W with no effects that I noticed. Just trying to figure out the bootable part, looking at systems like @kevinlekiller suggested.

kevinlekiller, 2020-11-08:
#cat /etc/systemd/system/cpuvoltage.service  
    [Unit]
    Description=Sets the voltage of the CPU.
    
    [Service]
    ExecStart=/usr/local/sbin/cpuvoltage.sh
    Type=oneshot
    Restart=no
    RemainAfterExit=yes

    [Install]
    WantedBy=multi-user.target

Start it with `sudo systemctl --now enable cpuvolate.service`

ulfw2020, 2020-11-08:
Thanks Kevin, so I assume I create a file in /etc/systemd/system called cpuvoltage.service? Then I paste those contents in and change the path of ExecStart to where my script is located? (/home/ulfwennstroem/undervoltScript) thanks.

ulfw2020, 2020-11-08:
I did the above, and ended up in a new situation. It doesn't work on boot, but if I run that systemctl command to enable it, it works fine. Any ideas?

kevinlekiller, 2020-11-08:
That's right, alternatively, if you put it in `~/.config/systemd/user/ ` and run `systemctl --user` with the rest of the commands, then the script will run when you log in with your user account.

kevinlekiller, 2020-11-08:
> I did the above, and ended up in a new situation. It doesn't work on boot, but if I run that systemctl command to enable it, it works fine. Any ideas?

It's probably a timing issue, you can either try like I mentioned above, a user unit or maybe put a sleep in the script to delay the execution.

kevinlekiller, 2022-03-13:
Not an issue anymore with 47001bfee5ec4b4ddddee686e4552382567268a9

closed 2022-03-13 by kevinlekiller