kevinlekiller/amdctl #14: Amdctl reports incorrect frequencies

echaskaris, 2019-06-13:
I have AMD FX-4170. Top frequency is 8400 mhz, but the CPU is rated for 4200.

kevinlekiller, 2019-06-14:
It might be setting the wrong FSB speed, some bulldozer CPU's are 100 MHz, some are 200, it multiplies the FSB by the multiplier to get the frequency (ex 42 * 100 = 4200).

kevinlekiller, 2019-06-14:
https://github.com/kevinlekiller/amdctl/blob/master/amdctl.c#L386

Is your CPU revision between 0 and 0f ?

echaskaris, 2019-06-14:
What field is the revision in /proc/cpuinfo?

echaskaris, 2019-06-14:
/proc/cpuinfo contents:
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 21
model		: 1
model name	: AMD FX(tm)-4170 Quad-Core Processor
stepping	: 2
microcode	: 0x600063e
cpu MHz		: 1630.357
cache size	: 2048 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 cx16 sse4_1 sse4_2 popcnt aes xsave avx lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs xop skinit wdt fma4 nodeid_msr topoext perfctr_core perfctr_nb cpb hw_pstate ssbd ibpb vmmcall arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold
bugs		: fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 8437.44
TLB size	: 1536 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 48 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate cpb

kevinlekiller, 2019-06-14:
Yeah, you're in between 0 and 0f (model	: 1), I wasn't sure about the math on those, because the manual says they have a FSB of 200 MHz, if you look through, search for cpudid / cpu fid / nbdid / nbfid / corecof,  you will find some explanations of how the CPU frequency is calculated, https://www.amd.com/system/files/TechDocs/42301_15h_Mod_00h-0Fh_BKDG.pdf

If I have time this weekend, I will look to see how it is done on these models.

kevinlekiller, 2019-06-14:
You could try commenting out this section https://github.com/kevinlekiller/amdctl/blob/master/amdctl.c#L390-L392 and see if any values look off.

echaskaris, 2019-06-14:
They look okay without that section. (4200 base, 4300 boost)

kevinlekiller, 2019-06-14:
Alright, the only thing I wonder is if it affects the north bridge clock calculations, maybe it requires the math on 200 MHz for north bridge calculations and 100 MHz for core clock calculations ?

Referring to this block of code I mean (which shows north bridge information - if refclck is now 100, do those numbers output wrong?): https://github.com/kevinlekiller/amdctl/blob/master/amdctl.c#L318-L328

kevinlekiller, 2019-06-14:
If so, this line might need to be changed: https://github.com/kevinlekiller/amdctl/blob/master/amdctl.c#L328

`(((cpuModel >= 0x00 && cpuModel <= 0x0f) ? REFCLK * 2 : REFCLK) * (nbdid + 0x4) / pow(2, nbfid))`

priand1, 2019-09-03:
Hi Kevin, thanks for amdctl. It works quite well. I also have the problem that echaskaris has, but with an FX-8320.
I was able to fix it by making the two changes mentioned above in amdctl.c:
[](https://github.com/kevinlekiller/amdctl/issues/14#issuecomment-502109322)
and
[](https://github.com/kevinlekiller/amdctl/issues/14#issuecomment-502113532)

My /proc/cpuinfo:
`processor       : 7
vendor_id       : AuthenticAMD
cpu family      : 21
model           : 2
model name      : AMD FX(tm)-8320 Eight-Core Processor
stepping        : 0
microcode       : 0x6000852
cpu MHz         : 1399.187
cache size      : 2048 KB
physical id     : 0
siblings        : 8
core id         : 7
cpu cores       : 4
apicid          : 23
initial apicid  : 7
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 popcnt aes xsave avx f16c lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs xop skinit wdt lwp fma4 tce nodeid_msr tbm topoext perfctr_core perfctr_nb cpb hw_pstate ssbd ibpb vmmcall bmi1 arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold
bugs            : fxsave_leak sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips        : 6999.84
TLB size        : 1536 4K pages
clflush size    : 64
cache_alignment : 64
address sizes   : 48 bits physical, 48 bits virtual
power management: ts ttp tm 100mhzsteps hwpstate cpb eff_freq_ro`

Thank you, Andreas

commit a6cc7b85 (2021-08-29): Some 16h CPU's have double the refclk speed #14

Closes #14
closed 2021-08-29 by kevinlekiller