kevinlekiller/amdctl #17: amdctl.service?

priand1, 2019-09-03:
Hi Kevin,

would you be interested in adding amdctl.service + .timer + amdclt-after-resume.service + example bash script to your package?
It is nothing big but I have written all of the above and they have been running on my system for a couple of weeks now without any problems.

kevinlekiller, 2019-09-06:
Sure, thanks for your work.

priand1, 2019-09-07:
Alright, here we go. Just to clear some things up:
I am on openSUSE Tumbleweed and the msr kernel module was not available early-on during boot, not even when including it in the initrd. Thus I decided to work around this by adding a systemd timer to execute amdctl via service 15s after boot.
15s after boot is sufficient on my system to have the msr module up and running when amdctl gets started.
For others 15s might be too short of a timespan (I am not sure about that). 
If you are sure your msr module gets loaded but have a message like the following in your journal when amdctl is started

"ERROR: Could not open CPU for reading! Is the msr kernel module loaded?"

just raise the value to 30s, 1min, 2min or whatever fits your needs.

The services and the script assume that amdctl resides in /usr/sbin.

A short Howto:
1. Copy the sample script to /usr/sbin, rename it (remove the .sample in the file name)  and make it executable.

2. Edit the script and fill in your tested values for voltage/clockspeed for each P-state.

3. Copy amdctl-boot.timer, amdctl-boot.service and amdctl-after-suspend.service to
/usr/lib/systemd/system

4. Enable amdctl-boot.timer with
systemctl enable amdctl-boot.timer

5. Enable amdctl-after-suspend.service with
systemctl enable amdctl-after-suspend

6. Reboot

You're done. 
From now on after every boot and after every wake-up from suspend-to-ram/hibernate/hybrid-sleep amdctl is run to apply your desired settings for voltage/clockspeed.

[amdctl-after-suspend.service.txt](https://github.com/kevinlekiller/amdctl/files/3586927/amdctl-after-suspend.service.txt)
[amdctl-boot.service.txt](https://github.com/kevinlekiller/amdctl/files/3586928/amdctl-boot.service.txt)
[amdctl-boot.timer.txt](https://github.com/kevinlekiller/amdctl/files/3586929/amdctl-boot.timer.txt)
[amdctl.sh.sample.txt](https://github.com/kevinlekiller/amdctl/files/3586930/amdctl.sh.sample.txt)

Note:
I had to add a *.txt ending to all files, otherwise github refuses to upload them. Just remove the .txt extension from the file names, please.

closed 2019-09-07 by priand1