kevinlekiller/amdctl #18: Cannot change CPU core voltage using amdctl

Mhury, 2019-09-26:
Hi all,

Recently, I have been testing this tool in my AMD Ryzen 5 PRO 2400G.
While I managed to change the frequency to overclock my CPU, I haven't been able to modify the CPU core voltage. For my experiments, I first disabled all pstates except for p0 and then I tried to change the VID corresponding to p0 to different values. Yet, using zenmonitor to measure the current CPU core voltage, I observed that these values were ignored by the CPU (they were completely different to the ones I saw in zenmonitor).

Do you have any idea why this tool cannot change the CPU core voltage in my machine?

```
cat /proc/cpuinfo 
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 23
model		: 17
model name	: AMD Ryzen 5 PRO 2400G with Radeon Vega Graphics
stepping	: 0
microcode	: 0x8101016
cpu MHz		: 3398.293
cache size	: 512 KB
physical id	: 0
siblings	: 8
core id		: 0
cpu cores	: 4
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb hw_pstate sme ssbd sev ibpb vmmcall fsgsbase bmi1 avx2 smep bmi2 rdseed adx smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves clzero irperf xsaveerptr arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif overflow_recov succor smca
bugs		: sysret_ss_attrs null_seg spectre_v1 spectre_v2 spec_store_bypass
bogomips	: 7186.16
TLB size	: 2560 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 43 bits physical, 48 bits virtual
power management: ts ttp tm hwpstate eff_freq_ro [13] [14]
```

```
sudo ./amdctl -c 0
Voltage ID encodings: SVI (serial)
Detected CPU model 11h, from family 17h with 8 CPU cores.

Core 0 | P-State Limits (non-turbo): Highest: 1 ; Lowest 3 | Current P-State: 1
 Pstate Status CpuFid CpuDid CpuVid CpuMult CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1    137      8      1  34.25x 3400MHz  1544mV     36     10  46.00A   71.01W
      1      disabled
      2      disabled
      3      disabled
Northbridge:
No P-States on AMD17H Northbridge.
```

```
cpupower frequency-info
analyzing CPU 0:
  driver: acpi-cpufreq
  CPUs which run at the same hardware frequency: 0
  CPUs which need to have their frequency coordinated by software: 0
  maximum transition latency:  Cannot determine or is not supported.
  hardware limits: 1.60 GHz - 3.60 GHz
  available frequency steps:  3.60 GHz, 2.30 GHz, 1.60 GHz
  available cpufreq governors: conservative ondemand userspace powersave performance schedutil
  current policy: frequency should be within 3.60 GHz and 3.60 GHz.
                  The governor "userspace" may decide which speed to use
                  within this range.
  current CPU frequency: Unable to call hardware
  current CPU frequency: 3.37 GHz (asserted by call to kernel)
  boost state support:
    Supported: no
    Active: no
```

[image]

I run "cpuid" and got the following extra info regarding the advanced power management features. 
Would it be possible that the tool doesn't work because the frequency ID (FID) control and the voltage ID (VID) control are set to "false"? 

```
 Advanced Power Management Features (0x80000007/edx):
     temperature sensing diode = true
     frequency ID (FID) control = false
     voltage ID (VID) control = false
     thermal trip (TTP) = true
     thermal monitor (TM) = true
     software thermal control (STC) = false
     100 MHz multiplier control = false
     hardware P-State control = true
     TscInvariant = true
```

kevinlekiller, 2019-09-26:
> Would it be possible that the tool doesn't work because the frequency ID (FID) control and the voltage ID (VID) control are set to "false"?

Sounds like it is, the voltage is changed using the "CpuVID" field per p-state.

Maybe another register needs to be modified to allow CpuVID to be changed.

It doesn't say anything on page 139 of Processor Programming Reference (PPR) for AMD Family 17h Models 00h-0Fh Processors (PUB), maybe somewhere else in the manual it states how to?

kevinlekiller, 2019-09-26:
I'm wondering if it's an issue with the CPU using the voltage from the turbo P-state (which I'm not sure how to modify, I don't think it's mentioned in the PDF). Maybe it only uses the P-state voltages if turbo is disabled or something like this?

Mhury, 2019-09-26:
Many thanks for your answer @kevinlekiller 

Following the manual you indicated, I read MSRC001_0015 (page 132) in my machine to verify if the turbo is enabled or disabled. This resulted in: 

```
sudo rdmsr 0xC0010015
b000011
```
Which means that the core performance boost (bit 25 starting from bit 0) is disabled. So in theory the CPU shouldn't be using the turbo P-states.

I've been trying to find a way to modify 0x80000007/edx to the activate frequency ID (FID) control and the voltage ID (FID) control, but I haven't figured out yet how to do so.

CHECK-1234, 2022-12-19:
@kevinlekiller @edumarinfabregas @ermo @theonlynexus @numas13

## SOLUTION FOUND
**Had the same issue:**
I was having this same issue where changing the voltage of a PSTATE does not work, even in the BIOS. Changing CPU voltage using Ryzen Master in Windows also did not work. 

**The Solution:**
The solution is to change your CPU CLOCK RATIO in BIOS to be above the base clock of your CPU.

**For Example:** 
I am using Ryzen 7 1700 where the base clock is 3.00ghz, in the BIOS, all I had to do was set the CPU CLOCK RATIO to 3.1ghz, when I rebooted back to the BIOS I noticed that the voltage was high (around 1.2v), then tried to change the voltage of the PSTATE in the BIOS. And it worked. To decrease the max frequency from 3.1ghz, all I did was set the Pstate0 CPU frequency to 2.5ghz.

**My Setup:**
Pstate0: 2.5ghz at around 0.768v
Pstate1: 1.55ghz at around 0.6v
Core Performance Boost: Disabled
C6 Residency: Enabled (Reduces idle power usage)
SOC VOLTAGE: around 0.68v (Reduces idle power usage)
Ram Speed: 1333mhz (Reduces idle power usage)

## **The new way to undervolt mobile Ryzen CPUs:**
https://github.com/DavidS95/Smokeless_UMAF

This allows you to access hidden BIOS settings which is good for computers with Ryzen mobile CPUs where the BIOS is locked.
This also works on Steam Deck, which you can now finally undervolt the Steam Deck.

EmbeddedCat, 2023-05-08:
Maybe a workaround in this problem on the code side is to adjust the cpu frequency slightly lower? again, if this an actual solution to the problem, it's not a very fancy solution, but it work hahaha

EmbeddedCat, 2023-05-08:
Hello everyone, I may have found a solution to this problem, this solution is not very fancy but it worked. It's something like what @CHECK-1234 said, but instead of changing the **CPU CLOCK RATIO** from the **BIOS** you changed it from a tool like this one or any other tool that can change the **CPU CLOCK RATIO**, like **zen-states**. In my case I used this tool to change the cpu clock ratio from **3.6Ghz** to **3.625Ghz**, I did this by changing the **FID** of the **pstate - zero**. After that I tried to change the **VID** value and actually worked. The cpu I used was the **Ryzen 7 1800x**, so I guess this method will work in any other model 1 family 17 processor, maybe.

w568w, 2023-08-19:
> **Undervolting Mobile Ryzen CPUs at BIOS-level:**
> 
> https://github.com/DavidS95/Smokeless_UMAF
> 
> This allows you to access hidden BIOS settings which is good for computers with Ryzen mobile CPUs where the BIOS is locked. This also works on Steam Deck, which you can now finally undervolt the Steam Deck.

Just a WARNING in case that you are not a careful reader:

Smokeless_UMAF **CANNOT** undervolt a **mobile** APU with integrated GPUs (usually in laptop / handheld) and might **BRICK** your device if you adjust `P0State Vid` in the UMAF, even for only +/-1 mV. **Do not try this unless you have a BIOS flasher (e.g. a BIOS chip programmer).**

Confirm that it will brick on a mobile Ryzen 7 4800H.