kevinlekiller/amdctl #22: how to change AMD multiplier on windows

gfanini, 2020-04-08:
could it be possible to compile for windows somehow, that I think windows or the cpu is using multipliers somehow below those specified by amdmsrtweaker for windows but it's really tricky 

https://github.com/mpollice/AmdMsrTweaker/issues/6

thank you for any possible clues

kevinlekiller, 2020-04-08:
I'm not sure, don't have much knowledge on Windows and don't have access to a computer with the Windows O/S on it, but if there's a way to read and write to MSR's on Windows then it is possible to port the source for sure.

gfanini, 2020-04-28:
Is it possible to modify north bridge multiplier for AMD family 15h ?
or alternatively somehow disable the lower frequency ones, to use say only P0 north bridge multiplier

closed 2022-03-13 by kevinlekiller