kevinlekiller/amdctl #31: Report AGESA version

drws, 2022-02-19:
There is a shortage of Linux tools that could report AMD's AGESA (_AMD Generic Encapsulated Software Architecture_) version. Windows-based [HWiNFO](https://www.hwinfo.com/) and [AIDA64](https://www.aida64.com/) reportedly [have the ability](https://www.reddit.com/r/Amd/comments/7mdn6n/how_to_identify_agesa_code_running_on_your_system/), but I couldn't find any Linux-based solution.

kevinlekiller, 2022-02-19:
I did a quick search in the Zen 3 PUB, but couldn't find anything to get the version, maybe there's something in some of the other documentation available?

Unfortunately, those Windows tools are all proprietary (CPU-Z included) ; no source code, impossible to know how they're doing it without reverse engineering which is not something I want to do.

kevinlekiller, 2022-03-25:
Going to close, since I can't find any information on how to get the AGESA version.

`inxi` shows a lot of info, but not the AGESA version: `sudo inxi -F`

closed 2022-03-25 by kevinlekiller

drws, 2022-04-15:
@kevinlekiller apparently more [information](https://bugzilla.kernel.org/show_bug.cgi?id=201473) was available all along. If you're still interested in this please also see [`RyzenAdj`](https://github.com/FlyGoat/RyzenAdj/issues/188) and [`inxi`](https://github.com/smxi/inxi/issues/172) issues.

drws, 2025-03-08:
@kevinlekiller please see [this comment](https://github.com/FlyGoat/RyzenAdj/issues/188#issuecomment-2661504880), an open-source tool for reading AGESA version has recently been published.