kevinlekiller/amdctl #32: Problem setting voltage {4700U}

moa5505, 2022-03-02:
Hello,
I might be missing something here (I am far from being an expert), but I tried to undervolt my processor and could not make it.
I am using a laptop with a 4700U (family 17h) with Linux Mint, I tried to execute the command `sudo ./amdctl -p0 -v104` and I got the following response: `mint@mint:~/amdctl$ sudo ./amdctl -p 0 -v 104 
Voltage ID encodings: SVI (serial)
Detected CPU model 60h, from family 17h with 8 CPU cores.
PREVIEW MODE OFF - P-STATES WILL BE CHANGED!

Core 0 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
 Pstate Status CpuFid CpuDid CpuVid CpuMult CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
ERROR: Could not write new value to CPU!
`

Does anyone know what I am missing ? 
Thank you !

kevinlekiller, 2022-03-02:
Perhaps you have writing to MSR disabled? If so, see https://github.com/kevinlekiller/amdctl#requirements

moa5505, 2022-03-03:
I thought this was only a problem with kernels > 5.9 
I am using Linux 5.4.0-91-generic x86_64

I tried `sudo modprobe msr` and got no message error

kevinlekiller, 2022-03-03:
Alright, not sure then.

moa5505, 2022-03-03:
I think I found the problem. I disabled Secureboot in the BIOS/UEFI and now it works with no problem !
I hope this will help somebody in the future :D 
You can close this. Thanks for your help and reactivity !

kevinlekiller, 2022-03-03:
Great, glad you got it sorted!

closed 2022-03-03 by kevinlekiller