kevinlekiller/amdctl #40: AMD Cesanne 5800H voltage control

weter11, 2022-04-01:
Hi! After some time of investigating problems, I found a way to coltrol frequency.
So changing boost by echo command not working, so I use Tuxedo control centre to do that and also for fan settings.
Don't know is it a program limitation, but I can't set CPUFid more than 127, so currently I change power states by:
```
**amdctl -p 0 -v 60 -d 4 -f 74

Detected CPU model 50h, from family 19h with 16 CPU cores (REFCLK = 100MHz ; Voltage ID Encodings: SVI (serial)).
Preview mode OFF.

Core 0 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
 Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1     74      4     60   37.00x  3700.00MHz  1175mV     32     10  42.00A   49.35W**
```

sysbench report that performance was changed compared to default 3200MHz max w/o boost, so values was accepted.
ryzen_monitor report
```
 CPU Model │         AMD Ryzen 7 5800H with Radeon Graphics │
│                           Processor Code Name │                                        Cezanne │
│                                         Cores │                                              8 │
│                                     Core CCDs │                                              0 │
│                                 Cores Per CCD │                                              8 │
│                                SMU FW Version │                                       v64.45.0 │
│                                MP1 IF Version │                                            v12 │
```
This program also report 3.7 GHz, but changing CpuVid has no effect at all, program always report Voltage from Core VRM 1.22-1.24V and power consumption 35-36W. I always tried set it to 4.0GHz at 0.8V, so it really don't change voltage for CPU.
So is it my settings incorrect, or there are other ways to change voltage on power states?

weter11, 2022-04-03:
Ok. So first some report from ryzen_monitor for cpu bench at 3200MHz:
```
╭─────────┬────────────┬──────────┬─────────┬──────────┬─────────────┬─────────────┬─────────────╮
│  Core 0 │   3200 MHz |  2.877 W | 1.059 V |  60.81 C | C0: 100.0 % | C1:   0.0 % | C6:   0.0 % │
│  Core 1 │   3200 MHz |  2.797 W | 1.059 V |  58.43 C | C0: 100.0 % | C1:   0.0 % | C6:   0.0 % │
│  Core 2 │   3200 MHz |  2.911 W | 1.059 V |  66.71 C | C0: 100.0 % | C1:   0.0 % | C6:   0.0 % │
│  Core 3 │   3200 MHz |  2.845 W | 1.059 V |  60.08 C | C0: 100.0 % | C1:   0.0 % | C6:   0.0 % │
│  Core 4 │   3200 MHz |  2.886 W | 1.059 V |  66.70 C | C0: 100.0 % | C1:   0.0 % | C6:   0.0 % │
│  Core 5 │   3200 MHz |  2.857 W | 1.059 V |  60.47 C | C0: 100.0 % | C1:   0.0 % | C6:   0.0 % │
│  Core 6 │   3200 MHz |  2.853 W | 1.059 V |  61.58 C | C0: 100.0 % | C1:   0.0 % | C6:   0.0 % │
│  Core 7 │   3200 MHz |  2.783 W | 1.059 V |  58.31 C | C0: 100.0 % | C1:   0.0 % | C6:   0.0 % │
╰─────────┴────────────┴──────────┴─────────┴──────────┴─────────────┴─────────────┴─────────────╯
╭── Core Statistics (Calculated) ───────────────┬────────────────────────────────────────────────╮
│              Highest Effective Core Frequency │                                       3200 MHz │
│                      Highest Core Temperature │                                        66.71 C │
│                          Highest Core Voltage │                                        1.059 V │
│                          Average Core Voltage │                                          inf V │
│                              Average Core CC6 │                                         -nan % │
│                          Total Core Power Sum │                                       22.809 W │
├── Reported by SMU ────────────────────────────┼────────────────────────────────────────────────┤
│                             Peak Core Voltage │                                        1.059 V │
╰───────────────────────────────────────────────┴────────────────────────────────────────────────╯
╭── Electrical & Thermal Constraints ───────────┬────────────────────────────────────────────────╮
│                              Peak Temperature │                                        68.25 C │
│                               GFX Temperature │                                        52.34 C │
│                         Voltage from Core VRM │               1.070 V |   1.450 V |    73.79 % │
│                                           PPT │              15.229 W |      54 W |    28.20 % │
│                                       PPT APU │              15.229 W |      54 W |    28.20 % │
│                                     TDC Value │              14.258 A |      58 A |    24.58 % │
│                           TDC Value, SoC only │               1.558 A |      15 A |    10.39 % │
│                                           EDC │              64.638 A |     110 A |    58.76 % │
│                                 EDC, SoC only │               0.000 A |      20 A |     0.00 % │
│                                           THM │               66.87 C |      98 C |    68.24 % │
│                                       THM SoC │               54.47 C |      98 C |    55.58 % │
│                                       THM GFX │               52.34 C |      98 C |    53.41 % │
│                                           FIT │                   0   |     260   |     0.00 % │
╰───────────────────────────────────────────────┴────────────────────────────────────────────────╯
╭── Memory Interface ───────────────────────────┬────────────────────────────────────────────────╮
│                                  Coupled Mode │                                             ON │
│                        Fabric Clock (Average) │                                       1600 MHz │
│                                  Fabric Clock │                                       1600 MHz │
│                                  Uncore Clock │                                       1600 MHz │
│                                  Memory Clock │                                       1600 MHz │
│                                     cLDO_VDDM │                                       0.9027 V │
│                                     cLDO_VDDP │                                       0.7514 V │
╰───────────────────────────────────────────────┴────────────────────────────────────────────────╯
╭── Graphics Subsystem──────────────────────────┬────────────────────────────────────────────────╮
│                       GFX Voltage | ROC Power │                          0.7107 V |    0.657 W │
│                               GFX Temperature │                                        52.34 C │
│                    GFX Clock Real | Effective │                           443 MHz |     88 MHz │
│                                      GFX Busy │                                       159.66 % │
│                     GFX EDC Limit | Residency │                          68.720 A |     0.00 % │
│                           Display Count | FPS │                                 1 |     0.00   │
│               DGPU Power | Freq Target | Busy │               0.000 W |     0 MHz |     0.00 % │
╰───────────────────────────────────────────────┴────────────────────────────────────────────────╯
╭── Power Consumption ──────────────────────────┬────────────────────────────────────────────────╮
[... cut ...]
│                              SoC Power (SVI2) │               0.944 V |   1.622 A |    1.531 W │
│                             Core Power (SVI2) │               1.226 V |  21.146 A |   25.915 W │
│                              Core Power (SMU) │                                       25.915 W │
│                            Socket Power (SMU) │                                       30.422 W │
│                           Package Power (SMU) │                                       30.422 W │
╰───────────────────────────────────────────────┴────────────────────────────────────────────────╯
```

kevinlekiller, 2022-04-03:
> Also when I try to set Fid more than 127

Ah! I see the issue, data type is signed, so it's limited to 127.

Will push a fix, thanks for the output!

weter11, 2025-11-20:
Technically, first should be hacked CPU microcode, then some changes to SMU, and after this voltage control is possible or at least changing offsets with CO. Another way is by changing BCLK CPU start to work at higher freq but at the same voltage, but I didn't find any tools for linux, https://github.com/sbski/Renoir-Mobile-Tuning can help for win users.
So, in theory its possible, but there no tutorials how to do all of this in reality. I can close issue if needed.

i2, 2025-11-20:
It makes sense. I was just curious. No need to close it as some work is needed. Thank you!