kevinlekiller/amdctl #44: Steam Deck: Pstate 0 Vid is 161, above max of 124

coolbho3k, 2022-07-07:
On the Steam Deck's custom AMD APU, the value of Pstate 0 is 161.

I'd like to undervolt this state! Would it be safe to raise the value of the `MAX_VID` macro to something a bit above 161?

Output: 
```
(deck@steamdeck amdctl)$ sudo ./amdctl -g
Detected CPU model 90h, from family 17h with 8 CPU cores (REFCLK = 100MHz ; Voltage ID Encodings: SVI (serial)).

Core 0 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
 Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1    112      8    161   28.00x  2800.00MHz   543mV     28     10  38.00A   20.63W
      1      1     96      8    114   24.00x  2400.00MHz   837mV     24     10  34.00A   28.46W
      2      1    102     12    105   17.00x  1700.00MHz   893mV     17     10  27.00A   24.11W

Core 1 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
 Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1    112      8    161   28.00x  2800.00MHz   543mV     28     10  38.00A   20.63W
      1      1     96      8    114   24.00x  2400.00MHz   837mV     24     10  34.00A   28.46W
      2      1    102     12    105   17.00x  1700.00MHz   893mV     17     10  27.00A   24.11W

Core 2 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
 Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1    112      8    161   28.00x  2800.00MHz   543mV     28     10  38.00A   20.63W
      1      1     96      8    114   24.00x  2400.00MHz   837mV     24     10  34.00A   28.46W
      2      1    102     12    105   17.00x  1700.00MHz   893mV     17     10  27.00A   24.11W

Core 3 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
 Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1    112      8    161   28.00x  2800.00MHz   543mV     28     10  38.00A   20.63W
      1      1     96      8    114   24.00x  2400.00MHz   837mV     24     10  34.00A   28.46W
      2      1    102     12    105   17.00x  1700.00MHz   893mV     17     10  27.00A   24.11W

Core 4 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
 Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1    112      8    161   28.00x  2800.00MHz   543mV     28     10  38.00A   20.63W
      1      1     96      8    114   24.00x  2400.00MHz   837mV     24     10  34.00A   28.46W
      2      1    102     12    105   17.00x  1700.00MHz   893mV     17     10  27.00A   24.11W

Core 5 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
 Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1    112      8    161   28.00x  2800.00MHz   543mV     28     10  38.00A   20.63W
      1      1     96      8    114   24.00x  2400.00MHz   837mV     24     10  34.00A   28.46W
      2      1    102     12    105   17.00x  1700.00MHz   893mV     17     10  27.00A   24.11W

Core 6 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
 Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1    112      8    161   28.00x  2800.00MHz   543mV     28     10  38.00A   20.63W
      1      1     96      8    114   24.00x  2400.00MHz   837mV     24     10  34.00A   28.46W
      2      1    102     12    105   17.00x  1700.00MHz   893mV     17     10  27.00A   24.11W

Core 7 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
 Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
      0      1    112      8    161   28.00x  2800.00MHz   543mV     28     10  38.00A   20.63W
      1      1     96      8    114   24.00x  2400.00MHz   837mV     24     10  34.00A   28.46W
      2      1    102     12    105   17.00x  1700.00MHz   893mV     17     10  27.00A   24.11W
```

coolbho3k, 2022-07-07:
Ah it appears that undervolting mobile Zen CPUs may not be possible even through the MSR. Please close if that's the case.  I tried manually tweaking the CpuVid and didn't seem to notice a difference.

CHECK-1234, 2022-12-19:
## **The new way to undervolt mobile Ryzen CPUs:**

**Undervolting Mobile Ryzen CPUs at BIOS-level:**
 
https://github.com/DavidS95/Smokeless_UMAF

This allows you to access hidden BIOS settings which is good for computers with Ryzen mobile CPUs where the BIOS is locked.
This also works on Steam Deck, which you can now finally undervolt the Steam Deck.

**You can also check:**
https://www.reddit.com/r/SteamDeck/comments/yq2yjl/how_to_overclock_your_steam_deck_fr/
https://www.reddit.com/r/SteamDeck/comments/zc58pt/ocuv_steam_deck_quick_guide_and_settings/

coolbho3k, 2022-12-19:
Yes, I know about Smokeless UMAF. It would be great to be able to control it from Linux, though.

Arian04, 2024-02-06:
just an FYI for anyone who comes across this with the same question I had. Yes, you can set `MAX_VID` to a higher value and build from source. I went through the AMD processor programming reference sheet for my cpu and couldn't find anything there or elsewhere online that indicated that I shouldn't set the core VID higher than 124. I'm guessing the author of this tool might've just thought it was a reasonable limit or something? My laptop hasn't exploded yet so I'm guessing there's no serious reason to not undervolt even harder if your hardware can handle it. If things go badly though, I'll try to remember to update this.