kevinlekiller/amdctl #54: Error writing to /dev/cpu/0/msr

Viciouss, 2023-08-01:
Hi,

I wanted to undervolt the 5300G in my homelab but no matter what I try, I always get:

ERROR: Could not write data to /dev/cpu/0/msr

msr module is loaded, I tried both the -m cli option as well as the sys fs parameter. I'm using kernel 6.4.6 currently. Output of -g is:

> Detected CPU model 50h, from family 19h with 8 CPU cores (REFCLK = 100MHz ; Voltage ID Encodings: SVI (serial)).
> 
> Core 0 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
> 
>  Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
>       0      1    160      8     32   40.00x  4000.00MHz  1350mV     40     10  50.00A   67.50W
>       1      1    102     12     96   17.00x  1700.00MHz   950mV     17     10  27.00A   25.65W
>       2      1     98     14    102   14.00x  1400.00MHz   912mV     14     10  24.00A   21.89W
> 
> Core 1 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
>  Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
>       0      1    160      8     32   40.00x  4000.00MHz  1350mV     40     10  50.00A   67.50W
>       1      1    102     12     96   17.00x  1700.00MHz   950mV     17     10  27.00A   25.65W
>       2      1     98     14    102   14.00x  1400.00MHz   912mV     14     10  24.00A   21.89W
> 
> Core 2 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
>  Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
>       0      1    160      8     32   40.00x  4000.00MHz  1350mV     40     10  50.00A   67.50W
>       1      1    102     12     96   17.00x  1700.00MHz   950mV     17     10  27.00A   25.65W
>       2      1     98     14    102   14.00x  1400.00MHz   912mV     14     10  24.00A   21.89W
> 
> Core 3 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
>  Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
>       0      1    160      8     32   40.00x  4000.00MHz  1350mV     40     10  50.00A   67.50W
>       1      1    102     12     96   17.00x  1700.00MHz   950mV     17     10  27.00A   25.65W
>       2      1     98     14    102   14.00x  1400.00MHz   912mV     14     10  24.00A   21.89W
> 
> Core 4 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
>  Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
>       0      1    160      8     32   40.00x  4000.00MHz  1350mV     40     10  50.00A   67.50W
>       1      1    102     12     96   17.00x  1700.00MHz   950mV     17     10  27.00A   25.65W
>       2      1     98     14    102   14.00x  1400.00MHz   912mV     14     10  24.00A   21.89W
> 
> Core 5 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
>  Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
>       0      1    160      8     32   40.00x  4000.00MHz  1350mV     40     10  50.00A   67.50W
>       1      1    102     12     96   17.00x  1700.00MHz   950mV     17     10  27.00A   25.65W
>       2      1     98     14    102   14.00x  1400.00MHz   912mV     14     10  24.00A   21.89W
> 
> Core 6 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
>  Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
>       0      1    160      8     32   40.00x  4000.00MHz  1350mV     40     10  50.00A   67.50W
>       1      1    102     12     96   17.00x  1700.00MHz   950mV     17     10  27.00A   25.65W
>       2      1     98     14    102   14.00x  1400.00MHz   912mV     14     10  24.00A   21.89W
> 
> Core 7 | P-State Limits (non-turbo): Highest: 0 ; Lowest 2 | Current P-State: 0
>  Pstate Status CpuFid CpuDid CpuVid  CpuMult     CpuFreq CpuVolt IddVal IddDiv CpuCurr CpuPower
>       0      1    160      8     32   40.00x  4000.00MHz  1350mV     40     10  50.00A   67.50W
>       1      1    102     12     96   17.00x  1700.00MHz   950mV     17     10  27.00A   25.65W
>       2      1     98     14    102   14.00x  1400.00MHz   912mV     14     10  24.00A   21.89W

cat /proc/cpuinfo:

> processor	: 0
> vendor_id	: AuthenticAMD
> cpu family	: 25
> model		: 80
> model name	: AMD Ryzen 3 5300G with Radeon Graphics
> stepping	: 0
> microcode	: 0xa50000c
> cpu MHz		: 2369.997
> cache size	: 512 KB
> physical id	: 0
> siblings	: 8
> core id		: 0
> cpu cores	: 4
> apicid		: 0
> initial apicid	: 0
> fpu		: yes
> fpu_exception	: yes
> cpuid level	: 16
> wp		: yes
> flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf rapl pni pclmulqdq monitor ssse3 fma cx16 sse4_1 sse4_2 movbe popcnt aes xsave avx f16c rdrand lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt tce topoext perfctr_core perfctr_nb bpext perfctr_llc mwaitx cpb cat_l3 cdp_l3 hw_pstate ssbd mba ibrs ibpb stibp vmmcall fsgsbase bmi1 avx2 smep bmi2 erms invpcid cqm rdt_a rdseed adx smap clflushopt clwb sha_ni xsaveopt xsavec xgetbv1 xsaves cqm_llc cqm_occup_llc cqm_mbm_total cqm_mbm_local clzero irperf xsaveerptr rdpru wbnoinvd cppc arat npt lbrv svm_lock nrip_save tsc_scale vmcb_clean flushbyasid decodeassists pausefilter pfthreshold avic v_vmsave_vmload vgif v_spec_ctrl umip pku ospke vaes vpclmulqdq rdpid overflow_recov succor smca fsrm
> bugs		: sysret_ss_attrs spectre_v1 spectre_v2 spec_store_bypass
> bogomips	: 7988.39
> TLB size	: 2560 4K pages
> clflush size	: 64
> cache_alignment	: 64
> address sizes	: 48 bits physical, 48 bits virtual
> power management: ts ttp tm hwpstate cpb eff_freq_ro [13] [14]

Any help is appreciated.

Kind Regards
Martin

AYuusuke, 2024-03-04:
Hi @Viciouss 

I had similar error because of UEFI+SecureBoot mode:
```
# sudo dmesg | grep -e secureboot -e amdctl
[    0.000000] secureboot: Secure boot enabled
[    0.014819] secureboot: Secure boot enabled
[85917.978818] Lockdown: amdctl: raw MSR access is restricted; see man kernel_lockdown.7
```
Legacy BIOS mode works fine for me. Does it help?

Viciouss, 2024-03-04:
Hey @AYuusuke 

That did the trick, thank you very much!

closed 2024-03-04 by Viciouss